NebraLtd/Helium-Guides #146: Support Ticket 265033

Kayz86, 2021-08-17:
Hi i apologise for creating this again, as my initial post was closed.

My Miner is not connected to blockchain after Nebra firmware update, please see my ticket: 265033 where I have submitted frequency and mac addresses.

Please also create a support thread so we can reach out to you appropriately.

Thank You

Kayz86, 2021-08-18:
My Outdoor Nebra has now become offline from the Helium network for the last 2 days. It is no longer mining after the most recent firmware update.

However the miner is online and is discoverable on the Nebra dashboard and my local network / nebra diagnostics page.

I have also restarted it to no avail.

Any help getting me back online will be much appreciated.

salmanfarisvp, 2021-08-18:
@Kayz86 I just responded to the support ticket, please note that these Github issues are for documentation related things,  even though the previous issue was closed, you can still mention it by posting queries there or you can follow up a support ticket.

closed 2021-08-18 by salmanfarisvp