NebraLtd/Helium-Guides #147: What's the difference between Reboot and Restart?

Kayz86, 2021-08-18:
So i am trying to fix my own miner, after firmware update I have been off the blockchain for 2 days my device has stopped mining.

I restarted my miner through the dashboard with no effect, I now just want to know if I reboot what will this do? Will I have to reassert my location and height etc?

salmanfarisvp, 2021-08-18:
@Kayz86 The Restart action **restarts the currently running application containers**. Your application's running containers will be removed and recreated from scratch. The Reboot action will do the entire device, including the kernel, which will be **rebooted as if there was a power cycle**.

closed 2021-08-18 by salmanfarisvp

shawaj, 2021-08-18:
@salmanfarisvp we should clarify this on the dashboard with a dashboard guide as in #67

Kayz86, 2021-08-21:
@salmanfarisvp 

If I do a reboot (power cycle) it will also restart the current running application containers as well? Basically the reboot will also perform the 'restart' mode as you've described it.

salmanfarisvp, 2021-08-22:
@shawaj Sure, I'm working on that. 

@Kayz86 Yes, correct.