NebraLtd/Helium-Guides #148: Syncing issue?

Aiva333, 2021-08-18:
Hello,

My miner is witnessing other miners, I got witnessed too, but on the Nebra dashboard, I'm still not fully synced? And the percentage is decreasing...

[image]

shawaj, 2021-08-18:
Can you please email sales@nebra.com with following info:

Model of unit?

Mac address of the unit (Shown as ETH on sticker)?

Frequency of the unit (Shown as Freq on sticker)?

How are you connecting it to internet? (Ethernet, Wi-Fi, Cellular)?

How are you powering the unit? (included adaptor, PoE, third party adaptor)

If the issue relates to initial setup of the hotspot, please also include:

What make & model of phone are you using?

What version of the Helium App are you using?

Do you have any screenshots of any error codes?

shawaj, 2021-08-18:
FYI sometimes percentage will go down if it is having trouble getting the next blocks from peers. You need to leave it for at least a few days

shawaj, 2021-08-20:
@Aiva333 please post the ticket number here once done. 

Closing this as it's not related to documentation issues.

closed 2021-08-20 by shawaj