NebraLtd/Helium-Guides #149: No onboarding key found

chipbrantley, 2021-08-18:
I know this is a common issue, but when I tried to connect my outdoor hotspot, I received a "No onboarding key found" message. I have emailed sales@nebra.com with the form filled out. Do I need to do anything else to report this issue? Thank you.

salmanfarisvp, 2021-08-19:
@chipbrantley Can you please share the ticket number? 

For now, I closing this issue, since it's not related to documentation, and you can share the ticket number in the same issue once you have it. Thank You.

closed 2021-08-19 by salmanfarisvp

chipbrantley, 2021-08-19:
@salmanfarisvp Thank you. The ticket number is 264746. -CB

chipbrantley, 2021-08-21:
@salmanfarisvp I noticed in another ticket that you advised keeping the hotspot plugged in and connected by Ethernet. Re: ticket # 264746, I have mine now plugged and connected. Could you tell me next steps please? Thank you!