NebraLtd/Helium-Guides #153: No onboarding key found

davidtopplayer, 2021-08-20:
Hello, i already wrote you many emails about my Problem (ID #260113), but you werent able to fix ist (you diednt answered me since 5(!!!) Days) and im getting frustrated. So Here i am.  Please Help me. I have the No onboarding key issued. I have attached a Screenshot. Please Help me.
[image]

davidtopplayer, 2021-08-20:
And yes, i already reinstalled the conpute modul

shawaj, 2021-08-20:
Please keep to the ticket as this is not related to the documentation.

Please also leave the miner on and connected to Ethernet.

And please make sure you have provided the following information on the ticket:
```
Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third party adaptor)
```

closed 2021-08-20 by shawaj