NebraLtd/Helium-Guides #154: Nebra indoor miner Green light off/still syncing

Spartan268, 2021-08-20:
Hello again, 

At present the green light on my miner has turned off. I read through the FAQs and everything seems to be fine on my end. Props to Aaron who got the miner syncing in the first place (thanks Aaron) Today it’s almost synced but with this issue I’m worried I’ll be back to square one soon. 

I can still run diagnostics via IP all ok and it’s showing all good but I’m not longer able to connect to Bluetooth to the Helium app. I’ve restarted the miner and my router to no avail. 

Can you please have a look into this? Is this a global issue? 

Thanks. 

Ticket #264141

shawaj, 2021-08-20:
The green light will only be on if you are in Bluetooth pairing mode

closed 2021-08-20 by shawaj

Spartan268, 2021-08-20:
Hey Aaron,

it won’t go into Bluetooth pairing mode when I press the button though. It’s just off all the time now.

Spartan268, 2021-08-20:
Also I don’t know if this is connected but it showing on the helium app that it hasn’t connected in 21 hours.

shawaj, 2021-08-20:
You need to hold the button down for about 10 seconds. We also had an issue that was fixed in today's update which caused the Bluetooth to sometimes not work. So you might want to try again