NebraLtd/Helium-Guides #156: Diagnostic Issue and syncing issues

Patchjay, 2021-08-21:
Hi
I have just come back from being away and have tried setting up my miners in a new location. They seemed fine where they were before for the few days I had them.

I am currently facing two issues.
I had synced the miner I have previously before in another location but have moved it down in London.

The Nebra Indoor is in syncing mode for a few days now, Helium App show that the hot spot is in syncing mode, but the IP diagnostic shows
[image]
[image]
[image]
[image]
 0% synced to the blockchain. The hardware lights are ok - green and amber lights are on.

As well as this when I try to go to the diagnostic page on the app I get this error attached. (last one)
I have tried rebooting and letting it sit for a few days but nothing is changing. 

Any help would be appreciated

Patchjay, 2021-08-21:
I included the mac address and all those other details in the support ticket if you need it here:
#267182
Thanks

vpetersson, 2021-08-21:
Closed and moved to support tickets.

closed 2021-08-21 by vpetersson