NebraLtd/Helium-Guides #158: Nebra Outdoor Hangs/Stalls often - Requires Reboot often?

Kayz86, 2021-08-22:
A few friends and I have a few nebra outdoor devices between us, I have just learned that one of the devices was just hanging with no activity going on?

**Device 1**

One of our devices was not performing for almost 24 hours, there was an odd activity of a challenge being created and that's it.

So after looking at the dashboard tonight, I noticed:

- CPU usage was 79%
- Memory was 529/962 mb
- Storage usage was 14.3 GB / 27.8 GB

I couldn't understand why the mine was so busy when it wasn't doing anything for the last 24 hours according to the Helium activity list?

So I decided to reboot it remotely. 

- Now the CPU usage is: 7%
- Memory usage: 574/962 mb
- Storage: 11.7 GB / 27.8 GB

After doing that the mine started to operate as normal, with the activity list growing every 20-30 minutes on the helium network. Just look at the huge CPU difference.

**Device 2**

This device has been fine, until today. I just noticed the activity list stopped 7 hours ago. Again I don't understand why there are no activities for such a long time?

Looking at the dashboard I suspect it's a similar issue as Device 1:

- CPU usage was 61%
- Memory was 584/962 mb
- Storage usage was 11.5 GB / 27.8 GB

Again CPU is high when it has not been doing anything for the past 7 hours?

I am now going to reboot Device 2.

**Device 3**

This device has had fewer issues than the other 2 devices, it also has a busy activity list without any huge breaks in between.

Why does the Nebra devices hang so quickly and in this way? Do I need to reboot the miners periodically for them to function properly?

vpetersson, 2021-08-22:
Please contact support@nebra.com

closed 2021-08-22 by vpetersson

Kayz86, 2021-08-22:
Although i have rebooted the devices and they appear to be working fine now, this was a generic question for discussion regarding why the devices hang like this.