NebraLtd/Helium-Guides #159: 4G LTE - No cellular connection

Almanaka, 2021-08-23:
Dear Nebra,

I am currently experiencing difficulties getting the LTE configuration to work for the Nebra Outdoor hotspot.

Have tested on two different miners with both Quectel EG25-G and EG25-G-S mPCIe modules. Have tested with SIM cards from two different network providers (TDC and 3 in Denmark) on both miners and manually added the APN details specified by said network providers.

Have also tried several other APN settings/configurations (broadband, mms, LTE, *99#, *99…1#, ect) not specified by network providers to perform general troubleshooting.

No configuration is currently supporting a connection via cellular connection. Outbound and inbound peer-to-peer connections reads as not-connected (gray x) in the Helium app for both hotspots

Connections via Wifi or Ethernet is working perfectly.

Modem Detected is “True” for Diagnostic Information for both hotspots.

SIM cards have been tested in Zyxel LTE4506-M606 mobile wifi unit and are working. SIM cards therefore not damaged or inactive. 

Outdoor Hotspot LTE configuration as per your guides has been followed.

Trying to determine source of problem:

1. Is any other action than pasting the cellular file into system-connection required? Your guide currently shows (you should also see some files already there called ???) which seems incomplete, so I am wondering if a step is missing? Does something need to be unplugged physically or changed when accessing resin-boot?

2. Does LTE antenna connection have an impact on cellular connection? No pictures of how to connect the connections cables is show in your current hotspot quickstart guide with regard to antenna placement. I have mine connected to the plugs on the outmost right and left. Not the middle. 

3. Is location assertion required on order for a cellular connection to be established?

Any assistance would be much appreciated.

Thank you.

Almanaka

shawaj, 2021-08-23:
How do you know it's not connected?

And are you sure your APN settings are correct? That's the most important part really. The hardware is working as it's detected by the miner.

Premmmi, 2021-08-24:
Hi, 

The miner was 100% synchronized with Blockchain, month back. Miner is 
connected via LAN, but do not light LAN diodes.  It also does not go through
the connect to Bluetooth button. Only under LED is shines.

Komu: NebraLtd/Helium-Guides ***@***.***>
Datum: 23. 8. 2021 23:26:17

"

(https://github.com/NebraLtd/Helium-Guides/issues/159#issuecomment-904144761)
"

shawaj, 2021-08-24:
@Premmmi this is unrelated to this issue. This issue is about a 4G connection

Premmmi, 2021-08-25:
Hi,

My hotspot suddenly stopped all activity, without ano logical reason...and 
they are dead for weeks now. They previously worked fine, my internet 
connection is top of the line, ports are open, the devices are backed up 
with colling units, everything should be in order...except it isnt, and my 
units arent mining at all. I ve manually unplugged and plugged them numerous
time...it didnt helped, restarted the routers...didnt helped. Ergo...I am 
here, asking you for your help and guidance.
I can provide you with additional information...MAC codes, ETH code and all 
that, just tell me where to approach you...since nebra support email doesnt 
give me any feedback for weeks now.

Can you help me, please?

Datum: 25. 8. 2021 1:03:10

"

(https://github.com/NebraLtd/Helium-Guides/issues/159#issuecomment-905032067)
"

zonfacter, 2021-09-02:
Hallo ich habe genau das gleiche Problem. Ich benutze auch die Quectel EG25-G.
Da als Benutzer keinerlei Möglichkeit besteht bei laufenden Hotspot eine 4G Verbindung zu prüfen - Terminal, Onboard Website. Auch ein Priorisieren einer Netzwerk Verbindung ist nicht möglich. Ich würde gerne weiterhin per WLAN oder LAN verbunden sein aber Trotzdem soll der Traffic über 4G laufen.
Auch ich habe die SIM Karte mit einem Externen LTE Router getestet wenn der Hotspot daran angeschlossen ist per LAN Funktioniert auch alles (Relay).
Mein Anbieter in Deutschland ist Mobilcom-Debitel - o2 Deutschland

**---- Google Übersetzt aus dem Deutschen -> Englisch**

Hi, I have exactly the same problem. I also use the Quectel EG25-G.
As the user has no way of checking a 4G connection while the hotspot is running - terminal, onboard website. It is also not possible to prioritize a network connection. I would like to continue to be connected via WLAN or LAN, but the traffic should still run over 4G.
I also tested the SIM card with an external LTE router if the hotspot is connected to it via LAN Everything works (relay).
My provider in Germany is Mobilcom-Debitel - o2 Germany

Almanaka, 2021-09-06:
@shawaj 

Thanks for your reply.

When rebooting the miner and running diagnostics, I get a gray x in inbound, outbound and sync status. When checking the data usage on the employed SIM card after a couple of days, no data transfers has been made. The wifi connection reads as not connected.

Honestly the APN settings might very well be the problem. At least I cannot rule out that my network providers has given me the wrong information. I will try with other providers and SIM cards in the following weeks in order to see this will resolve the matter. 

However, both my current network providers are quite confident that they have provided the correct APN setting. It is for this reason that I am also exploring other potential root causes for the missing cellular connection. Just in case I have missed or misunderstood something. :)

From your reply, I take it that none of the three questions I posted should have any impact on celluar connectivity? 

And you have not encountered anyone else that had to deviate from your LTE configuration guide to get cellular connectivity to work?

salmanfarisvp, 2021-09-13:
@Almanaka @Premmmi @zonfacter Please share the details with sales@nebar.com 

I'm closing this issue since it's not related to the documentation.

closed 2021-09-13 by salmanfarisvp