NebraLtd/Helium-Guides #161: Sudden loss of activity due to syncing issues

VictorHristov7, 2021-08-28:
Dears, 
It appears that with the latest firmware hotspots are losing Inbound connectivity or getting relayed out of nowhere which leads to loss of any activity and syncing.
Generally, it would be hard to catch it if it happens at night or when away from the device. 
It gets fix after at least few reboots - power off and on. There is no pattern when it goes out of connection - it might take 2 or 5 hours. Each time was different.

Some other observations with the latest updates of firmware:
1. Green light blinking eight times, stops for two seconds then starts again - wasn’t like that before the update
2. In Helium App Diagnostics - No IP is shown. Was not the case before the update. Still, it is ok on the Nebra Dashboard

I can see other users reporting this on several channels. One of them got a fix by the support team over the e-mail but if it is a general problem for all Indoor Devices, should we get a fix?

vpetersson, 2021-08-28:
Please contact support@nebra.com.

closed 2021-08-28 by vpetersson

sajmi, 2021-08-28:
I have the same problem