NebraLtd/Helium-Guides #163: Add more info in advanced diag for "green light is rapidly blinking"

jbnjohnathan, 2021-08-29:
**Is your feature request related to a problem? Please describe.**
As of the last few days all of my Nebra Wifi connected hotspots are offline.
The one I have at home only gets IP over Ethernet and not over WiFi.
I can see that the green LED is blinking rapidly about 10 times then stops, then does it again.
I found a section in the advanced troubleshooting guide for this behaviour, but no further information is listed there.

**Describe the solution you'd like**
I would like some troubleshooting steps for the issue with "green LED blinking rapidly" found here to be added: https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/advanced-diagnostics.md#green-light-is-rapidly-blinking

**Additional context**
Would also be good to explain that this means that WiFi cannot be used but Ethernet still works (if the blinking is only about that issue)

adam2740, 2021-09-04:
I am getting this problem aswell is there a solution provided to fix the problem?. I have reseated the usb components inside also plugged my hotspot up to a ethernet cable worked after syncing but when I go set it back up for WIFI it goes back to rapid green blinking and continuous syncing.  is it that the WIFI component inside the box is faulty?

DGMurdockIII, 2021-09-07:
I get this problem too I would love it to be fixed

Jayeshloveworld, 2021-09-07:
I am facing same problem too. Even though my hotspot is on continues on ethernet connection. However my online status is showing as mining and its not affecting my mining. Love to get it fixed if anyone have idea??

jaysaba, 2021-09-10:
I have a similar issue with my Nebra after update 2021.08.24.4..... and is still doing it now with the 2021.09.03.0 the green light keeps blinking and it mines properly for about 3 days straight and then it stops... it continues to show online for about a day but it does not mine and then it goes offline. I unplug it and plug it back.. it syncs and continues to mine for about 3 more days and so on... Please help me fix this!!!!! anyone else having the same issue???

vpetersson, 2021-09-11:
This is iot a support forum. Reach out to support@nebra.com

closed 2021-09-11 by vpetersson