NebraLtd/Helium-Guides #165: No Onboarding key

Nwilati, 2021-08-31:
Mac address of the unit (Shown as ETH on sticker): F0:4C:D5:AB:D6:B4 
•	Frequency of the unit (Shown as Freq on sticker): 868
•	How are you connecting it to internet? (Ethernet, Wi-Fi, Cellular): Eth
•	How are you powering the unit? Original power supply 
•	initial setup of the hotspot: N/A
•	What make & model of phone are you using? iPhone 11 
•	What version of the Helium App are you using? latest 
•	Do you have any screenshots of any error codes? Yes 

Syn to Helium network @100%

Mac address of the unit (Shown as ETH on sticker): F0:4C:D5:AC:75:E9
•	Frequency of the unit (Shown as Freq on sticker): 868
•	How are you connecting it to internet? (Ethernet, Wi-Fi, Cellular): Eth 
•	How are you powering the unit? Original power supply 
•	initial setup of the hotspot: N/A
•	What make & model of phone are you using? iPhone 11 
•	What version of the Helium App are you using? latest 
### device’s code does not exist on @ dashboard.nebra.com when adding a new device 
Syn to Helium network @100%

ticket ID #269571. 

[image]
[image]

Nwilati, 2021-08-31:
Can anyone please help me with the issue ? Thank you

Nwilati, 2021-08-31:
And I am very much aware that this issue is not related to documentation however I can’t find anyone from Nebra engineering team that would respond , so please let me know what to do , thank you

closed 2021-08-31 by salmanfarisvp

salmanfarisvp, 2021-08-31:
Please check the mail, shared the details. 

Closed this issue, since it's not related to documentation.