NebraLtd/Helium-Guides #166: Nebra disconnects / crashes after firmware update

Kayz86, 2021-08-31:
**Describe the bug**
After every update my one particular Nebra out of 3 Nebra devices crashes all the time.

Last time we fixed this as we identified it was due to a  Corrupted Blockchain Cache. I think Salman did a clean install.

Now I get more errors after the recent update:

- Region Plan: Node 'miner@127.0.0.1' not responding to pings.

- Miner Connected to Blockchain: False

- ECC Detected: False (This happened once before it became truel

**To Reproduce**
Steps to reproduce the behavior:

Send a firmware update to reproduce the error.

**Screenshots**
Attached from the hotspot diagnostics

**Desktop (please complete the following information):**
 - OS: Windows 
 - Browser Chrome
 - Version Latest

**Smartphone (please complete the following information):**
 - Device: Samsung
 - OS: Androis
 - Browser Chrome

**Additional context**

My Nebra has been offline for days. I have submitted a ticket but it has been pretty slow. Nebra needs more support workers please.
[image]

salmanfarisvp, 2021-08-31:
@Kayz86 Can you please share your previous ticket information? If you didn't have a ticket, please share the details with sales@nebra.com and share the ticket number here. 

I'm closing this ticket since it's not created to documentation.

closed 2021-08-31 by salmanfarisvp

Kayz86, 2021-08-31:
> @Kayz86 Can you please share your previous ticket information? If you didn't have a ticket, please share the details with [sales@nebra.com](mailto:sales@nebra.com) and share the ticket number here.
> 
> I'm closing this ticket since it's not created to documentation.

Hi Salman

I have been emailing all this time, the first ticket support email I sent did not give me a ticket number you just responded to my email. You have been referring to me as 'Kay'.

This email details were which does not have a ticket:

From: Nebra Support <sales@nebra.com>
Sent: 18 August 2021 15:44
To: (REMOVED)
Subject: Re: Blockchain: False after firmware update

Please also see tickets:

#268592
#265033

My email address begins with 'kayz' if you need to find it.

Thank you

Kayz86, 2021-09-03:
Guys Helium released a new firmware update to 2021.08.31.2 yet again for the 5th or so time my Nebra ceases to function and it requires Salman to do a clean install to fix the problem. It appears to be related to a Corrupted Blockchain Cache which relates to https://github.com/helium/miner/issues/792 https://github.com/helium/miner/issues/978

I also saw an ECC error relating to the memory of my miner on the diagnostics page, ECC Detected was showing as False before it went back to ECC Detected True.

The other outdoor miners i have purchased are performing fine except for this 1 particular device. I am losing out on numerous days of mining clearly this one is defective and i cannot keep asking support to fix this which can take up to a few days for anybody to action or respond.

I have a defective miner which I need replacing as soon as possible please.

DGMurdockIII, 2021-09-12:
I have same problem

DGMurdockIII, 2021-09-12:
ticket ID #273555