NebraLtd/Helium-Guides #168: Faulty Nebra Outdoor unit - needs replacing

Kayz86, 2021-09-04:
I have 4 outdoor devices, they all perform fine except 1.

1. Everytime there is a firmware update by Helium my device crashes and it requires for me to contact support to do a clean install. This has happened at least 3 times now consecutively.

2. Just 4 hours ago 11 pm (UK Time) my device crashed tonight and I could not access the diagnostics page. I then had to do a restart via the dashboard to get it back to some normality, but it keeps connecting and disconnecting from the blockchain.

I have sent numerous emails but it is falling on deaf ears, Salman appears to be manning the only support line all by himself. Last sent an email 2 days ago still have not had any support or response.

The Nebra device I have is faulty and I would like to have it replaced, please my order number is #2939 for your reference.

salmanfarisvp, 2021-09-04:
Please share hotspots details to sales@nebra.com. 

I'm closing this issue, since it's not related to documentation

closed 2021-09-04 by salmanfarisvp

Kayz86, 2021-09-04:
> Please share hotspots details to [sales@nebra.com](mailto:sales@nebra.com).
> 
> I'm closing this issue, since it's not related to documentation

Hi i had sent emails over 3 days ago no response, you can look at the most recent ticket which i have responded to almost 10 hours ago with my device details.

Ticket #272954

Thank you