NebraLtd/Helium-Guides #171: hotspot not in sync for over a week

dmtimdc, 2021-09-08:
**Describe the bug**
Since My Nebra indoor is down since aug/28/2021 23:57:21
Probably an update then?
The network interface went dead then.

On Monday aug/30 I power cycled the device and then it booted normally and the network seemed to be alive again.
But it seems not to come back in sync.

**Expected behavior**
Can you have a look at it? 
and provide me a solution, or an explanation, why?

**Screenshots**
<html>
<body>

Helium Address | View on Explorer
-- | --
Sync Percentage | 0.00%
Height Status | 0 / 1001730
Firmware Version | 2021.09.03.0
Frequency | 868
Region Plan | EU868
Variant | Nebra Indoor Hotspot Gen 1
Miner Connected To Blockchain | False

Miner Relayed | False
-- | --
ECC Detected | True
Ethernet MAC Address | 00:BD:27:78:0A:EB
WiFi MAC Address | 16:B7:56:58:9C:DC
Hardware Serial Number | 00000000ad516a59
Bluetooth Detected | True
LoRa Operational | True

</body>
</html>

**Additional context**
Add any other context about the problem here.

dmtimdc, 2021-09-08:
<html>
<body>

Sync Percentage | 0.00%
-- | --
Miner Connected To Blockchain | **True**
Some times **True** sometimes **false**

Height status 0/xxxxx
firmware version updates come true

</body>
</html>

vpetersson, 2021-09-08:
This is not a support forum. Please contact support@nebra.com.

closed 2021-09-08 by vpetersson