NebraLtd/Helium-Guides #178: Nebra Outdoor Miner: Storage almost full

slvcoin, 2021-09-11:
**Describe the bug**
I am operating my Nebra Outdoor Miner more than one month and it looks like the storage is filling up very quickly and it is almost full which can cause the miner stop working.
The storage is filled up by more than 80% (22.8 of 27.8GB are already used) and it increases by a minimum of 300MB per day, which means it will be full in approximately 2 weeks.
I can also not find any possibility to free up storage space and the miner is not doing it by itself.

**To Reproduce**
Operate the miner for a longer time period (more than 1 month)

**Expected behavior**
Storage space should not fill up at this speed or storage space should be freed from unnecessary files.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windowa 10
 - Browser Chrome
 - Version 92.0.4515.159

**Smartphone (please complete the following information):**
 - Device: n.a.
 - OS: n.a.
 - Browser n.a.
 - Version n.a.

**Additional context**
N.a.

KevinWassermann94, 2021-09-11:
The hotspot automatically checks the storage used on startup(reboot, power cycle or software update). If it is more than 80% it will automatically wipe the storage and perform a factory reset. So your hotspot will do a factory reset on the next startup.
It is not possible to just delete old blockchain data as it is still referenced by the newer blocks and will cause corruption of the data.

slvcoin, 2021-09-11:
If the miner wipes the complete memory and performs a factory reset, do I need to configure it again like during the first setup process or will the configuration be kept?

closed 2021-09-11 by slvcoin