NebraLtd/Helium-Guides #179: Add more info in advanced troubleshooting guide for "green light is rapidly blinking"

jbnjohnathan, 2021-09-11:
Creating this issue again as the previous one (https://github.com/NebraLtd/Helium-Guides/issues/163) was closed for incorrect reasons. I agree that this is not an support forum, and the comments made in the previous issue was not really belonging there, but my feature request is only about improving the documentation, which has not yet been completed.
Thus I am opening this issue again pending a merge of additional information in the documentation to describe the "Green light is rapidly blinking"

**Is your feature request related to a problem? Please describe.**
The section called "Green light is rapidly blinking" in the advanced troubleshooting guide is blank.

**Describe the solution you'd like**
I would like some troubleshooting steps for the issue with "green LED blinking rapidly" found here to be added: https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/advanced-diagnostics.md#green-light-is-rapidly-blinking

shawaj, 2021-09-12:
@salmanfarisvp is sorting this I believe.
But essentially it just means the following:

Blink fast if diagnostics are not OK (shows "errors found" in local diagnostics also)
Blink slow if advertising bluetooth
Solid if diagnostics are OK and not advertising

The 4 blinks on the blue led on daughterboard is usually a network connection issue.

jbnjohnathan, 2021-09-12:
> @salmanfarisvp is sorting this I believe.
> But essentially it just means the following:
>
> * Blink fast if diagnostics are not OK (shows "errors found" in local diagnostics also)
> * Blink slow if [...]

I have a unit that have the fast green blink. Doing a Diagnostic Report from the Helium app shows no errors.
Or is the "local diagnostic" something else? How can I access it?

I think it would be benificial if it was possible for the owner of the hotspot to better determine issues with it.

salmanfarisvp, 2021-09-13:
@shawaj Yes, I'm working on the documentation. 

@jbnjohnathan can you please share the hotspot details to sales@nebra.com , then share the ticket number here, will check and get back to you.

shawaj, 2021-09-13:
@jbnjohnathan The local diagnostics you can find here https://helium.nebra.com/handy-guides/local-diagnostics

jbnjohnathan, 2021-09-13:
> @jbnjohnathan The local diagnostics you can find here https://helium.nebra.com/handy-guides/local-diagnostics

Ah didn't know about this, thanks!
However my hotspot reports "All Ok" and still blinking : )

> @shawaj Yes, I'm working on the documentation.
> 
> @jbnjohnathan can you please share the hotspot details to [sales@nebra.com](mailto:sales@nebra.com) , then share the ticket number here, will check and get back to you.

will do!

jbnjohnathan, 2021-09-13:
@salmanfarisvp #275511

DGMurdockIII, 2021-09-21:
Please check my  ticket ID #273555

salmanfarisvp, 2021-10-19:
troubleshooting guide. https://helium.nebra.com/handy-guides/troubleshooting

closed 2021-10-19 by salmanfarisvp