NebraLtd/Helium-Guides #181: Improve structure / de-duplicate

shawaj, 2021-09-12:
**Describe the bug**

Right now we have duplication of lots of information. For example lots of troubleshooting info in the outdoor and indoor sections that are identical.

This is bad for a few reasons:

1.  It is confusing to have the same info in lots of places
2. It adds significant extra overhead to maintain the same information in several places
3. It is very likely that we will forget to update in all locations due to (2) above meaning that we will have differing information in multiple places, adding further confusion.

This is already happening in several places and will only get worse as the amount of content increases.

**Expected behavior**

We should improve the structure of the documentation so that we only have information in a single place about a single subject and don't duplicate... If we want to have the information in multiple places we should provide internal links rather than duplicating.

For example, this can be achieved in the troubleshooting pages by having:
- overall troubleshooting
- indoor troubleshooting
- outdoor troubleshooting
- light troubleshooting
- etc etc for new devices

The main troubleshooting can then link to the sub sections and the sub sections back to the main troubleshooting. In the main one should be troubleshooting relevant to all products, in the sub sections should be ones specific to that variant.

The same goes for the FAQ, onboarding and various other sections that are common to all devices.

This will be a major improvement and set us up nicely for the future.

salmanfarisvp, 2021-09-13:
@shawaj Thanks for raising the issue and I agree with you. 

Like you said we make a common troubleshooting guide, then reference all the troubleshooting links on that page section depends on the device with an internal link.  so the structure will be the same, but there will be only one main page for all the troubleshooting.

salmanfarisvp, 2021-09-14:
Here is the sample main troubleshooting page where all the troubleshooting related things listed and device-specific troubleshooting will be documented on the device directly and it will be redirected from the main troubleshooting pagehttps://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/troubleshooting.md 

also, each specfic device troubleshooting has a contents list, so user can navigate easily https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/troubleshooting.md . 

For common troubleshooting guides like "sync issue" ..etc will be added under the main troubleshooting guide add link to the device-specific contents list. 

@shawaj please check and let me know your comments.

salmanfarisvp, 2021-09-14:
Added content page here for easy navigation: https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/overview.md 

Removed other antenna docs from the sidebar and added content page https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/antenna/overview.md