NebraLtd/Helium-Guides #182: Off grid / renewable power

shawaj, 2021-09-12:
We should add some info about off grid considerations and link to guides / info / solutions for this kind of thing.

For starters, @bigdaveakers has a nice guide we could link to https://www.instructables.com/Helium-Hotspot-Off-Grid-Solar-Set-Up-for-Nebra-Out/

salmanfarisvp, 2021-09-13:
@shawaj can we create a new **resource** category under the handy guides, then we can list this information and their details?

shawaj, 2021-09-13:
Yeah good plan 👍

zonfacter, 2021-12-29:
Hello, I have also built an outdoor version and I have to say that this is not practical in autumn / winter / spring in Europe - Germany ... Since the consumption of 3 kWh per month cannot be covered with a small battery storage (100ah 12V I have max.  5 days buffering).  with 2x 130W solar panels and MPPT solar controller.  Since the weather here is just 10-14 days cloudy.  Even with a 4.5 KwP system, a charge of just 0.1 KWh comes together!  If you want to buffer this time of year, the investments are far too high, what you have to pay for battery storage.  A 100W panel is sufficient in summer.

salmanfarisvp, 2022-02-09:
Thanks for sharing @zonfacter, Do you have write-ups or photos you can share with us?, We will display them on our website under your name. Thanks.

salmanfarisvp, 2022-02-09:
@shawaj Created a resource page https://github.com/NebraLtd/Helium-Guides/commit/eb2096705ff483507eb3c203aefd9ec8b703a949 for sharing helium related resources and community projects and added to the sidebar under handy-guides.

salmanfarisvp, 2022-02-25:
Added @bigdaveakers  projects under **Helium Resource and Community Projects**.  https://helium.nebra.com/handy-guides/resource

closed 2022-02-25 by salmanfarisvp