NebraLtd/Helium-Guides #186: Nebra Dashboard

Kayz86, 2021-09-18:
Hi when the Nebra Dashboard launched it was working, now any commands sent to any of my miners it does not work. If I need to reboot or shut down i am having to do it manually.

When will we see a fix for the dashboard please?

salmanfarisvp, 2021-09-19:
Please share the details with the error code to sales@nebra.com

closed 2021-09-19 by salmanfarisvp

Kayz86, 2021-09-19:
There are no error codes, it executes a restart or a shutdown, yet 24 hours pass nothing has happens. This is across all 3 of our devices in different locations with different setups.

Before when we execute a command it use to happen within minutes or within 5 minutes. Now nothing, forced to manually switch the device off.

Also emailing takes over 1 week for somebody to respond....