NebraLtd/Helium-Guides #188: Indoor miner no earnings past 7 days

Spartan268, 2021-09-22:
Hi there,

I’m opening a ticket here as my support tickets for the last little while have been closed without action. 

I need assistance, my Nebra indoor miner is not earning and despite saying it’s online it’s not. Can you please review my ticket and let me know if you require any other information?
Ticket (#278681)

Thanks!

closed 2021-09-22 by vpetersson

Spartan268, 2021-09-22:
Hey V,

Can you please have a look at my ticket?

Shujasheikh, 2021-09-22:
Hi,

Check the prove of coverage plus depending on your setup and how many
miners Round where you based. We people establishing helium network to buy
these devices, once your hotspot performing properly will get rewards.

Or if find yourself fed up with nebra free contact to sale off.

Thanks
Shuja