NebraLtd/Helium-Guides #190: 3 devices not syncing / offline after update

Kayz86, 2021-09-30:
I have always had this problem since purchasing 3 outdoor Nebra devices. Purchased back in February 2021 delivered in the last week of July 2021.

All my Nebra devices go offline after an update or do not sync or come off the blockchain.

I have had a lack of support and now it takes almost 3 weeks to receive a response. If i create a thread here it gets closed pretty quickly.

Due to the lack of support I try and fix it myself. I turn the devices off for an hour and restart my internet. This usually clears things up. However with the most recent Helium update (27th Sept) its been 2 days I have not been able to fix anything.

All 3 devices in 3 different locations are either syncing or are stuck or offline.

I have emailed both sales and support and have created a ticket #281540 please respond soon, do not take 3 weeks. 🙏🏽

There should be a section here for support so people can also try and help one another.

salmanfarisvp, 2021-10-01:
Sorry to hear the issue and apologies for the delayed response, Just now responded via mail and please don't create many tickets. 

**This issue is not related to documentation, so closing this issue.**

closed 2021-10-01 by salmanfarisvp

Kayz86, 2021-10-04:
> Sorry to hear the issue and apologies for the delayed response, Just now responded via mail and please don't create many tickets.
> 
> **This issue is not related to documentation, so closing this issue.**

Salman I responded to you 4 days ago, again no response. I have 4 devices with the same issue so I doubt it has anything to do with the internal parts. I cannot show you all them as some are fitted really high above.
​
The issue related I think I have is related to a Corrupted Blockchain Cache, please see:

https://github.com/helium/miner/issues/792 https://github.com/helium/miner/issues/978

Can you please help me?

Kayz86, 2021-10-05:
Hi i am still facing issues, no help from anybody. A lot of people are having problems, our miners are going offline from the network whilst they appear to be fine on diagnostics.

Will this ever be resolved? I get the feeling Nebra are aware of the issues but are not willing to fix this. All our devices have not been mining for a very long time still. What are we to do or where to go?