NebraLtd/Helium-Guides #191: lora module false after 210927 snapshot updated

hollejacklin, 2021-10-01:
I got my indoor miner 210927, when it power on the green led blinking fast 5-6 times repeatedly, but the local diagnostic showed all ok, and starting syncing. After snapshot updating, the lora module false occurred. I try to cross tested the lora module in my another new indoor miner. I uninstalled the lora module from synced miner, and installed to new miner which didn’t syncing. The green led blinking slowly. It shows the lora module is ok.
I also try to install the lora module from unsynced miner to synced miner, and the green led blinking fast, and diagnostic shows lora false. I seems some bug cause by snapshot updated. How can solve this problem?
I ticket is #281657, and sent you the diagnostic file.

salmanfarisvp, 2021-10-01:
Sorry to hear the issue and apologies for the delayed response, Just now responded via mail and please don't create many tickets.

This issue is not related to documentation, so closing this issue.

closed 2021-10-01 by salmanfarisvp

Shujasheikh, 2021-10-02:
Hi 

Hotspot added into basket cannot process payment issue