NebraLtd/Helium-Guides #204: Hotspot needs attention

StretchZa, 2021-10-08:
Hello I received my nebra hotspot on Monday set it up through wifi it was online and syncing for about 3 days and then out of nowhere it just went offline and not syncing. I tried everything they mentioned on discord and the blogs, tried restarting it, rebooting it, unplugging it and plugging it back in. I opened it up maybe something was loose that wasn't the case. I Checked the port forwarding it's all fine. I connected it through Ethernet now but still same issue. It's connected only to the internet. I know a lot of people just say to wait but I don't think that would help me cuz it's not syncing so what am I waiting for. I can't run diagnostics through the app cuz it won't let me pair it through Bluetooth I keep on running to the same issue as well as typing the IP address to the browser it gives me no information on my hotspot. Ticket #283963 if anyone can help I'd really appreciate it! My hotspot's name is cheerful boysenberry porpoise
[image]
[image]
[image]

Shujasheikh, 2021-10-08:
Hi, NebraLtd/Helium-Guides

What are you expecting from Resberry PI? Be patience will take time not
forever.

Thanks

shawaj, 2021-10-08:
Looks like the issue is the storage being full. Did you buy this from someone second hand?

StretchZa, 2021-10-08:
I did email nebra support but they haven't responded to me.

StretchZa, 2021-10-08:
Ticket #283963