NebraLtd/Helium-Guides #222: Nebra Dashboard - Unable to remove Nebra Indoor

hunet-hr, 2021-10-31:
Please can anyone help with removing an Nebra miner from my Nebra Dashboard, 
as I no longer own the device & the new customer is unable to register it onto there own Nebra Dashboard ?
Message: _"The device has already been claimed"_

Kayz86, 2021-11-01:
> Please can anyone help with removing an Nebra miner from my Nebra Dashboard, as I no longer own the device & the new customer is unable to register it onto there own Nebra Dashboard ? Message: _"The device has already been claimed"_

that's bad.... the Nebra dashboard has been very buggy. In fact it doesn't work or do anything for me anymore.

salmanfarisvp, 2021-11-01:
We are working on the feature, now the user can't remove the device from his dashboard! if you want the device to be removed? please share the below details to sales@nebra.com for further investigation?

Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?

I'm closing the issue since it's not related to documentation.

closed 2021-11-01 by salmanfarisvp