NebraLtd/Helium-Guides #223: 500 Internal Server Error on Outdoor miner

Kayz86, 2021-11-01:
This has started to happen to my outdoor Nebra recently, it's happened twice now. On the first occasion I did a power cycle and everything came back to normal. However this has happened again for the second time in 2 days and it is still not coming out of the: "500 Internal Server Error" message.

See below it is the error I am getting when I try and reach the diagnostics page, however when I check my router's control panel I can see the miner's IP address as connected.

Please help

Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

salmanfarisvp, 2021-11-01:
Sorry to hear that, can you please share the below details to sales@nebra.com for further investigation? 

Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to the internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third-party adaptor)

I'm closing the issue since it's not related to documentation.

@Kayz86 please don't open any issues here that are not related to documentation, as you know we are managing all the support over mail, I hope you understand this.

closed 2021-11-01 by salmanfarisvp

Kayz86, 2021-11-01:
> Sorry to hear that, can you please share the below details to [sales@nebra.com](mailto:sales@nebra.com) for further investigation?
> 
> Model of unit (Indoor or Outdoor) Mac address of the unit (Shown as ETH on sticker) Frequency of the unit (Shown as Freq on sticker) What country are you using the miner in? How are you connecting it to the internet? (Ethernet, Wi-Fi, Cellular) How are you powering the unit? (Included adaptor, POE, Third-party adaptor)
> 
> I'm closing the issue since it's not related to documentation.
> 
> @Kayz86 please don't open any issues here that are not related to documentation, as you know we are managing all the support over mail, I hope you understand this.

Hi Salman I have sent this information through email however I know it can take 3 weeks for a response. I apologise for posting here.

I have sent all the information in my ticket #290726

Thank you

Kayz86, 2021-11-10:
Hi @salmanfarisvp @shawaj 

Despite emailing you 10 days ago with all the required information I still have not had 1 response from anybody.

I am not able to access my diagnostics page as I keep seeing an internal server error. I have to power cycle it almost everyday to get it to work. I feel like this device is dead now.

How can I get it replaced or repaired? I would like to bring my device down to your office to fix as I fear sending it in via post would just mean my miner will get lost and I will not hear from anybody.

salmanfarisvp, 2021-11-10:
@Kayz86 Sorry for the delay, Customer support will get back to you with a resolution.

Kayz86, 2021-11-17:
> @Kayz86 Sorry for the delay, Customer support will get back to you with a resolution.

@salmanfarisvp what action will you be taking now? Please dont blame this issue on the Helium network as your colleagues have done because I've been reporting it for months. So many other users are also complaining in discord.

Despite the Helium issues I am able to access the diagnostics page for other devices except for 1. I can also access my Bobcat fine without issue.

Today I am faced with the same problem on the same device. I have been power cycling it every few hours for the past 24 hours and I still cannot access the diagnostics page.

Your colleague confirmed via email this a 'software' issue apparently and he claimed to had fixed it. However it is clear nothing has been fixed.

I've had enough of the lack of support going to and fro with no end or resolution. I would like this device replaced. Please only contact me on how I can get a replacement outdoor device.