NebraLtd/Helium-Guides #225: Migrate to GitHub Pages

vpetersson, 2021-11-02:
The setup should be fairly straight forward:

* Create a GitHub Actions flow that builds the [Docsify](https://docsify.js.org/#/) static pages
* Serve them over [GitHub Pages](https://pages.github.com/). They will be served on https://nebraltd.github.io/Helium-Guides/ for now, but we will switch the domain over later.

After we're done, we need to make sure that no links changed from the old vs new setup.

## References

* [Easy way to create your Documentation site with Docsify and GitHub Pages](https://bryantson.medium.com/easy-way-to-create-your-documentation-site-with-docsify-and-github-pages-d4868b5b56e9)
* [How to create a documentation site with Docsify and GitHub Pages](https://opensource.com/article/20/7/docsify-github-pages)

vpetersson, 2021-11-02:
@salmanfarisvp Feel free to take a stab at this. Don't think it will be too complicated to be honest.

salmanfarisvp, 2021-11-02:
@vpetersson  Right now the https://nebraltd.github.io/Helium-Guides/ page is served from the **pages** branch, right? So I should modify that and make it work and check the URL first right? 

Thanks for the resource!

vpetersson, 2021-11-02:
No, I changed it to 'main':
[image]

shawaj, 2021-11-02:
The only things we really need to do is add some redirects and some auto cache purging that is now possible with our Cloudflare enterprise account

shawaj, 2021-11-04:
Added cache purging https://github.com/NebraLtd/Helium-Guides/commit/06b95889604b3e1d2554f99315a8ec44f6897698

Now just need to add redirects from the old domain structure to the new one. Can probably be done with a Cloudflare page rule I think

shawaj, 2021-11-04:
FYI @vpetersson docsify is not a true static page generator. This is why the current setup does not work properly on pages.

What we need to do, after we have set up the page redirects, is change the setup back to hash routing:
https://github.com/NebraLtd/Helium-Guides/blob/5211ebebaad4b89a4053e96e6d734188ae33fa19/docs/index.html#L57

This will allow it to work on pages

Alternative is to move to a true static site generator but that is probably one for the future IMO

shawaj, 2021-11-04:
Looks like static is on their roadmap for v5 https://github.com/docsifyjs/docsify/issues/1061

shawaj, 2021-11-04:
For reference https://docsify.js.org/#/deploy?id=github-pages

shawaj, 2021-11-04:
i just need to add the redirects and then we should be ready to switch over

You can see it at https://nebraltd.github.io/Helium-Guides/#/

shawaj, 2021-11-04:
@vpetersson actually - the one thing i am not sure of with GH Pages is serving the PDFs.

Need to check how we are currently doing this on the VPS

vpetersson, 2021-11-04:
> @vpetersson actually - the one thing i am not sure of with GH Pages is serving the PDFs.
> 
> Need to check how we are currently doing this on the VPS

This is done using [this script](https://github.com/NebraLtd/Helium-Guides/blob/main/pdf-gen/gen-pdfs.sh). That should be easy enough to incorporate in GitHub Actions.

shawaj, 2021-11-04:
Ah ok thanks @vpetersson will try and get a github action set up for that 👍

vpetersson, 2021-11-04:
> Ah ok thanks @vpetersson will try and get a github action set up for that 👍

Let @salmanfarisvp take a stab at that instead.

shawaj, 2021-11-04:
For reference https://github.com/pandoc/pandoc-action-example

shawaj, 2021-11-05:
having trouble setting up the redirects...

[image]

shawaj, 2021-11-05:
Been experimenting also with cloudflare pages https://646ee11a.helium-guides.pages.dev/

salmanfarisvp, 2021-11-05:
@shawaj Did you switch the GitHub page serving to "page" branch? I'm testing the links now and I can see a few of them are broken and I'll fix it.

shawaj, 2021-11-05:
@salmanfarisvp Yeah I changed it to pages branch just for testing so as not to break the existing site before switchover

If you have found links that aren't working, best thing is probably to create a draft PR to main branch with the updates which we can just merge when we switch to pages

shawaj, 2021-11-05:
In fact @salmanfarisvp maybe just add them to the PR #228 on pages branch

salmanfarisvp, 2022-02-09:
@shawaj Internal and external links are checked and working fine.

shawaj, 2022-02-25:
Fixed in https://github.com/NebraLtd/Helium-Guides/commit/9c26c8e8cf2ea9ca84fc4d219a556f28d791ea90

closed 2022-02-25 by shawaj