NebraLtd/Helium-Guides #232: Nebra miner constantly syncing and going offline

bennibry, 2021-11-05:
I have had my miner for a month now , first 2 weeks were fine but now its constantly going offline and trying to sync. I have contacted support but I have heard nothing from them
Diagnostics say it's fine, I have had it plugged into the ethernet to try and get it synced faster. Is there anyway to hard reset it ?

Thanks

[image]
m

bennibry, 2021-11-05:
[image]

salmanfarisvp, 2021-11-05:
Sorry to hear that, can you please share the below details to sales@nebra.com for further investigation?

Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to the internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third-party adaptor)

I'm closing the issue since it's not related to documentation.

closed 2021-11-05 by salmanfarisvp