NebraLtd/Helium-Guides #234: Link to PDF versions of files

shawaj, 2021-11-07:
After moving to github pages as part of #225 we will lose the ability to directly link to the PDF directory at https://helium.nebra.com/pdfs

So we need to create a PDF overview page at https://nebraltd.github.io/Helium-Guides/#/pdfs/overview.md or similar which links to the raw PDFs.

We could also add the raw PDF links to the main pages as well for reference purposes

ProTechEx, 2021-11-20:
Link is dead, as like your support !

salmanfarisvp, 2022-02-09:
Added PDF docs in the overview file https://github.com/NebraLtd/Helium-Guides/commit/221193c18c27d47bb8c61c7f78333bc53553c8eb  and listed on the sidebar https://github.com/NebraLtd/Helium-Guides/commit/5bf02f8225d5b934b99da9e49ce902dbfd96a408

salmanfarisvp, 2022-02-17:
Link https://helium.nebra.com/pdfs/overview

closed 2022-02-17 by salmanfarisvp