NebraLtd/Helium-Guides #237: Storage purge and api listener up address

andy210000, 2021-11-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hi

my Nebra indoor miner had a storage purge about 10hrs ago and since then the hotspot is now at block 1 of the blockchain sync and is not moving ? Can you please advise how I resolve this?

separate issue is the public IP address of my device is not the same as the listener api address which is causing issues with witnessing my and sending beacons . Please advise how to fix this as a reboot of the device has not resolved this

hotspot name ambitious umber mule

salmanfarisvp, 2021-11-10:
Sorry to hear that, can you please share the below details to sales@nebra.com for further investigation?

Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to the internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third-party adaptor)

I'm closing the issue since it's not related to documentation.

closed 2021-11-10 by salmanfarisvp

andy210000, 2021-11-10:
hi @salmanfarisvp i have but like all support calls never get a response please reopen this