NebraLtd/Helium-Guides #240: Miner stuck loading

prandash, 2021-11-10:
My miner 
Eth Mac F0:4C:D5:58:E6:AB
Rpi serial 00000000b45cf0ff
Is on a public IP address is reporting miner still loading been like this for week or Better. 
[image]

salmanfarisvp, 2021-11-10:
Sorry to hear that, can you please share the below details to sales@nebra.com for further investigation?

Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to the internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third-party adaptor)

I'm closing the issue since it's not related to documentation.

closed 2021-11-10 by salmanfarisvp

prandash, 2021-11-10:
I sent this all to sales already along with the diag file.

prandash, 2021-11-10:
---------- Forwarded message ---------
From: Nebra Support ***@***.***>
Date: Tue, Nov 2, 2021, 9:11 AM
Subject: Ticket (#291228) Received (PLEASE READ) - Additional info
To: ***@***.***>

Dear Phillip Randash,

We would like to acknowledge that we have received your request and a
ticket has been created with ticket ID #291228.

At the moment we're experience a big surge in customer support tickets, so
if you have emailed us for any reason then please note that it may take a
bit longer than usual to respond. *Do not email us multiple times as this
will slow down our response time.*

*Please read this if you're emailing about the Nebra HNT Miners *

At the moment we're experience a big surge in customer support tickets, so
if you have emailed us for any reason then please note that it may take a
bit longer than usual to respond. *Do not email us multiple times as this
will slow down our response time.*

We'd kindly like to ask anyone planning to email us regarding shipping, or
questions surrounding the miners to *please read through our FAQ page first*
. 90% of the queries can be answered via our FAQ guides
<https://helium.nebra.com/FAQs>.

It includes details such as:

   - What is the difference between the Indoor & Outdoor Units?
   <https://helium.nebra.com/FAQs?id=what-is-the-difference-between-the-indoor-amp-outdoor-units>
   - What frequency do I need for my country?
   <https://helium.nebra.com/FAQs?id=what-frequency-do-i-need-for-my-country>
   - When will the Nebra Hotspot ship?
   <https://helium.nebra.com/FAQs?id=when-will-the-nebra-hotspot-ship>
   - and much more

Also.....

   - Check our Maker Zone <https://www.pi-supply.com/make/> area which
   provides support material for our products. (e.g. Pi PoE, PaPiRus, Gert VGA
   666, Bright Pi and others)
   - Take a look at our Pi Supply YouTube channel
   <https://youtube.com/PiSupply> where we have a number of unboxings,
   tutorials and video guides
   - If you have troubles with your JustBoom boards please check our FAQ
   section <https://www.justboom.co/faqs/> and technical section
   <https://www.justboom.co/technical-guides/>
   - Finally much of our code and technical material can be found on our Pi
   Supply <https://github.com/PiSupply> and Nebra
   <https://github.com/NebraLtd> GitHub repositories
   - For StakeBox customers, you can visit our Learning section
   <https://www.stakebox.org/blogs/learn> here
   - You can also contact us via our Discord channel
   <https://pisupp.ly/chat>

Thank you for your patience.

Sincerely,

Nebra Ltd Support Team

[image: Pi Supply] <https://www.pi-supply.com/>

*NEW!* We now offer a number of customisation / kitting services. Click
here to learn more. <https://pisupp.ly/services>

291228:220300