NebraLtd/Helium-Guides #241: Miners no longer syncing since last night

FrothyJack, 2021-11-10:
Mine and several others online are experiencing an issue where the miners have stopped syncing.

Some have reset to block height 1 and others just have stopped at their current block.

[image]

FrothyJack, 2021-11-10:
looks to be the same issue sensecap is reporting 

It is reported that a small part of SenseCAP M1 has been updated the "LoRa region name" (e.g. updated from US915 to region_us915), if the M1 has been rebooted after the activation of PoCv11, which may result in the region unidentifiable issue. This new firmware will fix this issue.

my region plan is "region_eu868"

salmanfarisvp, 2021-11-10:
Sorry to hear that, can you please share the below details to sales@nebra.com for further investigation?

Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to the internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third-party adaptor)

I'm closing the issue since it's not related to documentation.

closed 2021-11-10 by salmanfarisvp