NebraLtd/Helium-Guides #248: Nebra Green Status LIght Not Working Cant Pair

NeoA83, 2021-11-16:
Unplugged and re plugged in miner.  Was all setup now the status light will not come on cant get into diagnostic through IP or app.  How do i troubleshot if i cant access router?  Have a ticket in with Nebra but it is taking awhile.  Any thoughts?

Sir1670, 2021-11-16:
I am having the same issue. When I try to connect to the Hotspot, i get the following error: Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

rebooted numerous times to no avail.

JesseFlyz, 2021-11-17:
I’m having the same issues. Can’t pair new miner with helium app. What to do next

NeoA83, 2021-11-17:
Also getting internal error when putting in the IP all ports are open and it was just working. I have a ticket in with Nebra can't find any info on how to fix this. Wish it could be reset. Took the whole thing apart re sat all components still nothing very frustrating. Hopefully someone will have an answer

shawaj, 2021-11-17:
Email support@nebra.com

P.S. there is a major Helium Network outage which is probably the root cause... see status.helium.com

closed 2021-11-17 by shawaj