NebraLtd/Helium-Guides #249: Internal Server Error The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

Sir1670, 2021-11-16:
The green light on the back of the unit does not come on, and when i try to connect to it via a web browser, I get the following message: Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

I have rebooted numerous times, I have double checked to ensure that all components within the unit are properly seated. All other lights seem to be functioning correctly (the 4 inside as well as the yellow on the back).

Fast Vanilla Bat is the Hotspot name.

shawaj, 2021-11-17:
Email support@nebra.com

P.S. there is a major Helium Network outage which is probably the root cause... see status.helium.com

closed 2021-11-17 by shawaj

Kayz86, 2021-11-17:
@shawaj its not fair to pass this issue on as a Helium problem. The Helium issue is new. I have been having this problem for almost 2 months with no fix provided from yourselves at Nebra.

Here is my thread https://github.com/NebraLtd/Helium-Guides/issues/223

@salmanfarisvp even said that an agent will be allocated. The Agent contacted me almost 2 weeks later to tell me he had resolved it as it was a software issue.

I have several Nebra devices and also Bobcat. Of all the Nebra devices this one is showing the Internal Server Issue. Very unfair of you to pass the buck and say its Helium issue. Theres hundreds of others complaining of the same issue on Discord.

So when will this be addressed?