NebraLtd/Helium-Guides #252: Update Indoor Troubleshooting guide

KevinWassermann94, 2021-11-17:
**No lights are turning on:**
- add information to check that the compute module is properly seated (add pictures)
- try a different power adaptor if available 
- wait for the hotspot to boot up
- on first startup wait for the hotspot to update

**Hotspot is not showing in the Helium App** 
- wait for the hotspot to boot up before pairing

**Hotspot is not detecting Wi-Fi networks**
- Only 2.4G Wifi is supported

**Syncing Issues**
- We provide an instant sync feature that download a new snapshot on startup
- If you ever find your hotspot more than 300 blocks behind according to the local IP diagnostics you should reboot the hotspot

**Add Onboarding Key Issues**
- Please contact our tech support via sales@nebra.com and provide your MAC address. We will update the onboarding key for you

**Add Lora concentrator false**
- Reseat Lora concentrator if it is showing for the first time(with power turned off)
- Contact support if that doesn't resolve it

**Add Hotspot not witnessing**
- check your antenna connector/ cable
- Improve the antenna placement/ place it as high as possible

ProTechEx, 2021-11-20:
I have the issue that i have 0 witness and the hotspot doesn't beaconing. Before 1st Nov everything has work fine. 
I had around 50 witnesses, before 1st Nov, currently I should have a more because there a lot of new hotspots nearly. 

The Hotspot is mounted on roof of building, arround 50m up from the ground with 360 degrees free sight. 

I have check out again for wetness penetration and and whether the antenna is no longer strongly in place. But everything is okey without any issues.

I have read that many others had this problem after the firmware update, so it can only be related to a buggy firmware.

Your support doesn't reply my ticket since 14 days. 

for me one thing is certain I will never buy Nebra hotspots again!

salmanfarisvp, 2022-01-20:
Added missing informations : see https://github.com/NebraLtd/Helium-Guides/pull/306

closed 2022-01-20 by salmanfarisvp