NebraLtd/Helium-Guides #254: Failed connection to Helium

JWomm, 2021-11-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Helium Dashboard
2. Click on a Hotspot
3. See error Failed connection to Helium
.
**Expected behavior**

.so im not able to shut miner down  - see status ect

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - android newest chrome

**Smartphone (please complete the following information):**
 - s10 galaxy

**Additional context**
Add any other context about the problem here.

salmanfarisvp, 2021-11-18:
Sorry to hear that, can you please share the below details to sales@nebra.com for further investigation?

Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to the internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third-party adaptor)

I'm closing the issue since it's not related to documentation.

closed 2021-11-18 by salmanfarisvp

shawaj, 2021-11-18:
I believe this is due to issues with Helium API being unreliable ATM