NebraLtd/Helium-Guides #256: Nebra outdoor hotspot.  internal server error for dashboard and wifi daughterboard will not work

jaw2ay, 2021-11-19:
**Describe the bug**
After the halt, my miner started acting up.  when I plugged in the nebra outdoor through POE the wifi daugherboard will no longer light up.  bluetooth does not work, wifi does not work,  and I am receiving a 500 internal server error when I try to hit the local dashboard.  all other lights are working.  I have reseated all of the components and believe this is a software issue.

**To Reproduce**
Steps to reproduce the behavior:
1. plug in nebra outdoor miner through POE
2. wait for it to fully boot
3. check wifi dauterboard light, it is not on
4. go to nebra local troubleshooting page, 500 internal server error

**Expected behavior**
wifi and bluetooth should work as expected and I should be able to press the button to pair.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: ios, windows
 - Browser [e.g. chrome]
 - Version [e.g. 22]

jaw2ay, 2021-11-19:
[image]

pic of the unresponsive daughterboard

St0rmz1, 2021-11-19:
I am experiencing the same behavior. After the blockchain halt, my nebra hotspots were able to install the new software and successfully synced with the blockchain. Roughly 8 hours later, I discovered all my hotspots exhibiting this same behavior. Both indoor and outdoor units.

shawaj, 2021-11-19:
Please give them some time to recover from the block halt. The network is still recovering

Additionally this is not a support forum. For support please email support@nebra.com

closed 2021-11-19 by shawaj

umuttcannn, 2021-11-23:
I have the same problem, my bluetooth light hasn't come on for 2 weeks, I haven't been able to connect since I bought it.