NebraLtd/Helium-Guides #257: Nebra indoor hotspot green led doesn't come on

LucioPro, 2021-11-19:
**Describe the bug**
Only the yellow LED comes on when I plug in the hotspot
I cannot get the green led to come on even when I hold in the physical button for 15 seconds
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

shawaj, 2021-11-19:
Please give them some time to recover from the block halt. The network is still recovering

Additionally this is not a support forum. For support please email support@nebra.com

closed 2021-11-19 by shawaj

LucioPro, 2021-11-19:
Thank you very much for your help. My apologies for the misuse of the
forum. I am still learning.