NebraLtd/Helium-Guides #268: Outdoor Hotspot stuck syncing

rgrego90, 2021-11-24:
Nebra Ticket (#315785)
Since Nov 21st my hotspot has been syncing (stuck on that blockchain). 
It has received the new image 2021.11.22.0 as soon as it was released to nebra hotspots but that has not changed anything. 
HeliumStatusio sees my device online, nebra dashboard sees my device online and synced with 0 witnesses.
Helium sees my witnesses but shows my hotspot syncing, my hotspot diagnostic page shows its syncing, went from %99.76 to %99.75 today it was around %99.50 (going in the wrong direction)

shawaj, 2021-11-24:
Please email support@nebra.com

closed 2021-11-24 by shawaj

rgrego90, 2021-11-24:
I did email nebra support, hence the Nebra Ticket (#315785). You guys don't respond.

rgrego90, 2021-11-26:
Please respond to my nebra ticket.