NebraLtd/Helium-Guides #276: Ethernet lights keeps repeatedly going on and off.

fishplayhard, 2021-12-03:
My Nebra hotspot Ethernet lights keeps repeatedly going on and off.

It was working fine before.

Does anyone has any leads on this issue?

salmanfarisvp, 2021-12-03:
Sorry to hear that, can you please share the below details to sales@nebra.com for further investigation?

Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to the internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third-party adaptor)

I'm closing the issue since it's not related to documentation.

closed 2021-12-03 by salmanfarisvp

Robotpants22, 2021-12-05:
I am having the same issue any luck

salmanfarisvp, 2021-12-05:
@Robotpants22 Please share the above-mentioned details with customer support.