NebraLtd/Helium-Guides #279: eMMC not recognized

AmitHayherYYC, 2021-12-06:
**Describe the bug**
my hotspot stopped working after 2 weeks and has been unable to connect via ethernet, wifi or bluetooth. TOook out he SD card and plugged into a PC. it is dead and not recognized.

What are the steps of getting a new SDflashed with the correct image for my hotspot

vpetersson, 2021-12-06:
This is not a support forum. Please contact [support@nebra.com](mailto:support@nebra.com).

closed 2021-12-06 by vpetersson

system-zero37, 2022-01-21:
Is there documentation on how to flash a new SD card @vpetersson ??  2 out of 3 of my miners are dead and it looks like its the SD card. Emailed support about this over a week ago and no response, I guess thats why every 2nd thread here is a support issue.

vpetersson, 2022-01-21:
No, you can't reflash the unit or replace any components yourself. That would break the warranty.