NebraLtd/Helium-Guides #286: Change Sx1301_hal to 1302_hal

serbyxp, 2021-12-17:
My origonal concentrator card broke on me, I switched to sx1302, I know the HAL is diffrent and the reset triggers are different. I have the miner hooked up to my laptop and have access to the file system. I created a settings.toml file in the network manager folder with the two sx1302_hal references in the hm-packet forwarder but I had no luck still says “lora operational false” the green light is on the new sx1302 card. I tested the card on regular raspi and it does work. I just need to figure out how to update the docker env in the emmc card

serbyxp, 2021-12-19:
Still haven’t been able to load new firmware that supports sx1302_hal it’s a v1 miner, so it doesn’t have the proper triggers / ENV settings, I have looked for 3 days now trying out how to get the ENV changed but I am having no luck. Any help please

shawaj, 2021-12-20:
We do not support this kind of modification I'm afraid.

Having said that, our miners support sx1302 based concentrators with no modifications necessary. So long as they have the same pinout as ours

closed 2021-12-20 by shawaj