NebraLtd/Helium-Guides #287: No Challenges Being Received by Miner

ebjorklu, 2021-12-18:
I'm trying to figure out why I haven't sent any beacons in the last week.  (I'm witnessing and sending challenges fine.)

Port forwarding is the most obvious issue, but I've:
1) restarted the miner multiple times; and 
2) checked the local web-app to verify miner is NOT in relay (diagnostic file attached); and
3) telnetted to my miner on port 44158 from outside my firewall successfully (which seems to further validate I'm not in relay).

Here's the URL to my miner on Helium Explorer if helpful.

https://explorer.helium.com/hotspots/112eupWoT1DnvwEi3pDM3v7xai9tmcsdG5EsQnXBfiXJjKyNNNZ

Let me know if there is any other information I can provide.

Thanks

-Eric

[nebra-diag.json.txt](https://github.com/NebraLtd/Helium-Guides/files/7739654/nebra-diag.json.txt)

ebjorklu, 2021-12-18:
I'm curious if there is any way to determine if I'm not:
1) receiving the challenges; or
2) receiving the challenges, but unable to send a beacon?

Let me know if there's any way to determine that from the diagnostic file I sent, or if something further needs to be pulled from the miner (i.e. additional logs).

Suprman0, 2021-12-19:
Same
https://explorer.helium.com/hotspots/11qyCRDpGSGdywxuM5P4QJNzCNDYAf7W62iKpiLycd62NbBNLgS/activity

marketstorm1, 2021-12-19:
I have the same problem too. No beacons sent for over 7 days and no witnesses.  I seem to be witnessing others as normal but I lost all of my witnesses. Many people in my area look like they are having the same problem
https://explorer.helium.com/hotspots/112iXx8mHSMbFDRapuj7yUAqM8BhA55GnBr3fyzezoSkV5hbpp81/activity

Suprman0, 2021-12-19:
Here are 7 more individuals.
https://www.reddit.com/r/HeliumNetwork/comments/riq3tp/nebra_miners_and_beaconing_since_pocv11/?utm_medium=android_app&utm_source=share

curtcormier, 2021-12-19:
Same here dropped to zero witnesses on both Outdoor Nebra's.

shawaj, 2021-12-20:
This is not a support channel.

Please email support@nebra.com

closed 2021-12-20 by shawaj