NebraLtd/Helium-Guides #288: Hotspot not sending beacons

marketstorm1, 2021-12-20:
Seems that many including mine have not sent a beacon for over a week and all is well with network connections and local troubleshooting at our hotspot locations.
Firmware Version 2021.12.14.0-1
Make: Nebra
Helium Explorer activity link :
https://explorer.helium.com/hotspots/112iXx8mHSMbFDRapuj7yUAqM8BhA55GnBr3fyzezoSkV5hbpp81/activity

shawaj, 2021-12-20:
You beaconed 41 minutes ago...

[image]

closed 2021-12-20 by shawaj

marketstorm1, 2021-12-20:
Hi... Just saw that on my end too. Thanks

shawaj, 2021-12-20:
No probs! We pushed an update yesterday evening which should have helped with this 👍