NebraLtd/Helium-Guides #295: Awaiting Location Assertion

tdurnan, 2022-01-01:
**Describe the bug**
status of awaiting location assertion and Lora Operation FALSE

All other statuses in Diagnostic show good and synced to network,\.
Location New Hamburg Canada

**To Reproduce**
Is status on startup

Have reseated the LORA card and antenna connections with power off.  Restart system and LED1 status light is clear for 1-2 min and then starts blink red 7-9 flashes hard to count.

shawaj, 2022-01-02:
This just means you haven't set up the hotspot yet and haven't asserted it's location.

It will show that until you assert location during the app and the lora will not work because it doesn't know what location or frequency to use.

If you have any further issues please email support@nebra.com

closed 2022-01-02 by shawaj

tdurnan, 2022-01-03:
Thanks I got it sorted out. Helium app needed to be done again. Not sure why but running now.   Do you know the max Dbi antenn the outdoor miner. CAn drive?  I live in a rural area and earnings have been very low