NebraLtd/Helium-Guides #296: Debug miner using serial interface

Underlyingglitch, 2022-01-06:
**Is your feature request related to a problem? Please describe.**
I have about 5 hotspots who show the "LoRa ready: False" error on the local diagnostics page. Some of these miners are fully synced to the blockchain and others aren't (this could however just be an issue with patience). I removed and installed the LoRa module several times and checked the entire inside of the miner for any clues. I found none.

**Describe the solution you'd like**
For people that are technical, it would be really nice to offer some tools to debug and potentially fix the issue themselves. Support is slow because many people are having issues and need help. For those who could do a big part themselves it would be really nice to have the documentation and tools to do so. I spotted several options for connecting external devices such as a Micro USB connector and a serial interface, but data seems to come from this.

**Describe alternatives you've considered**
I tried taking apart some parts from the miner and carefully putting everything back together, but nothing seemed to help.

**Additional context**
[image]

Underlyingglitch, 2022-01-06:
EDIT: perhaps the issue wasn't named correctly. It isn't specifically about the serial port. I would like to be able to debug the miner in any way possible. The diagnostics page just tells me something is wrong, I would like to know what.

Is it for example possible to access logs? Restart services manually? Or maybe just freshly pull the docker images? As far as I know, all miners run the software found here: https://github.com/NebraLtd/helium-miner-software/blob/master/docker-compose.yml

shawaj, 2022-01-06:
It shows lora operational false because you haven't asserted the location yet.

We will not be adding serial debug.

If you need any further assistance please email support@nebra.com

closed 2022-01-06 by shawaj

KevinWassermann94, 2022-01-06:
@Underlyingglitch You are able to restart the services through our dashboard https://dashboard.nebra.com/

Underlyingglitch, 2022-01-06:
> It shows lora operational false because you haven't asserted the location yet.
> 
> 
> 
> We will not be adding serial debug.
> 
> 
> 
> If you need any further assistance please email support@nebra.com

I can't add a location yet since the miner is "not ready yet". I have added other miners that show "All okay" before adding a location.

This is clearly not the issue

Underlyingglitch, 2022-01-06:
> @Underlyingglitch You are able to restart the services through our dashboard https://dashboard.nebra.com/

@KevinWassermann94 Thank you for the reply. The dashboard will unfortunately become a paid product, which means I'm trying to avoid it. Besides, that won't solve my issue since the miner isn't connected to Helium yet, which results in this error:
[image]

shawaj, 2022-01-06:
This is not a support channel.

Please email support@nebra.com

shawaj, 2022-01-06:
And just to further clarify, you are incorrect. It will always show lora operational false if you haven't asserted the location yet. 

Because the concentrator can't start up until it knows what region it's in.

Underlyingglitch, 2022-01-08:
> And just to further clarify, you are incorrect. It will always show lora operational false if you haven't asserted the location yet. 
> 
> 
> 
> Because the concentrator can't start up until it knows what region it's in.

@shawaj 
I'm incredibly sorry! You are correct. After setting a location the error disappeared. I have no idea why other hotspots showed the green "All ok" even when no location was set but I'm glad this fixed it.

As for your other comment. On your website you listed this repo (and specifically the issues page) as an alternative way to reach you. https://helium.nebra.com/support

I thought it would be okay for me to use this. It usually takes really long before an email is replied to and often the contents of the reply aren't about the question that was asked. I really liked the fact that I got a fast and meaningful reply here on GitHub just as I loved this when Helium still allowed manufacturers to use their Discord server for that purpose. I have seen how you guys were treated there so I understand why they stopped it. I have however always tried to be nice to other people, especially people that try to help me.

I will stop using GitHub for questions, but I do advice you to remove it from the webpage listed above as that makes it really confusing!

Have a nice day!