NebraLtd/Helium-Guides #298: Outdoor Miner

Stormyblues, 2022-01-06:
**Is your feature request related to a problem? Please describe.**
What to do when Bluetooth doesn’t work on setup and Button doesn’t reset.

**Describe the solution you'd like**
How to start the miner without Bluetooth access.  Using power not POE.  

**Describe alternatives you've considered

looked at troubleshooting page and it talks about going online.  Unit isn’t accessible yet pages indicate going to the internet.  How do I access my outdoor unit if I haven’t been able to set it up yet.  

**Additional context**
Add any other context or screenshots about the feature request here.

KevinWassermann94, 2022-01-06:
Please send a mail to sales@nebra.com, this is not a support channel

closed 2022-01-06 by KevinWassermann94

Stormyblues, 2022-01-06:
The sales channel is days behind answering anything.  This is something which should be in the setup manual.