NebraLtd/Helium-Guides #302: Quectel EC25-AU 4G modem

system-zero37, 2022-01-11:
I have installed the quectel EC25-AU modem into my outdoor miner. I followed the steps as described in the documentation but can not get the miner to work via 4G.

I have paid extra for a static IP with the SIM card. When I put the SIM into my phone, it connects to the internet as expected. I recorded the APN details that worked in my phone and added them to the cellular file required in the miner but still no dice. 

The modem section on the diagnostics page shows it as installed.

Is there any way I can debug and see what the 4G module is doing? Any further troubleshooting I can try?

**To Reproduce**
Steps to reproduce the behavior:
1. Use Optus SIM card (Australia)
2. Install Quectel EC25-AU 4G Modem
3. Edit cellular file with APN details that work in phone.

**Screenshots**
[image]

[image]

[image]

**Additional context**
Add any other context about the problem here.

shawaj, 2022-01-11:
Also are you sure the APN is correct?

Says here different ones https://www.optus.com.au/for-you/support/answer?id=1379

system-zero37, 2022-01-13:
I had the antennas connected when I was testing. The blank lines were not an issue. I had the .txt removed from the file. The issue may have been poor 4G signal inside my building, but that's hard to know without being able to see any debug showing what the modem is doing.

I got it working but still have issues with the miner being relayed. I put the sim card in my phone and found the IP address to be a public address (49.199.xxx.xxx) I'm not sure how the miner can be relayed considering there's no router in the way to block any ports.

The 2 APNs I have found to be working are "yesinternet" and "yesbusiness"

Is there a way to check the miners IP address?

shawaj, 2022-01-19:
Yes you can see the public IP from the Nebra dashboard https://dashboard.nebra.com

It's unlikely you will get it unrelayed on a mobile connection though as they use CGNAT and not a fixed IP address

shawaj, 2022-01-19:
Closing as resolved

closed 2022-01-19 by shawaj