NebraLtd/Helium-Guides #307: Risk of Hack - New hotspot transfer feature

Almanaka, 2022-01-19:
Just read you new update regarding the ability to transfer hotspots to another user via your dashboard. Eventhough this is a nice feature, I am personally very concerned that such a feature has been introduced to your online dashboard for which hackers can gain easy access via only login and password.

For the Helium app itself, a hacker will need to get their hands on my phone in order to perform a transfer. This is not the case with the dashboard portal.

Please consider introducing the option for two-factor authentication asap for both log-in to the dashboard and for performing the hotspot transfer. Hotspots are currently at risk.

shawaj, 2022-01-19:
Hotspots are not at risk because this doesn't transfer ownership on the blockchain it only transfers it between two different dashboard accounts

For feature requests for dashboard please post those here ... https://github.com/NebraLtd/dashboard-issues/issues

shawaj, 2022-01-19:
Additionally if you use the Google or GitHub social login then you can already use 2FA through them

closed 2022-01-19 by shawaj

Almanaka, 2022-01-19:
Thanks for the elaboration. Makes sense.