NebraLtd/Helium-Guides #312: Hola, hello, he testeado con éxito el modem modem 4G EG25-G-S

yamilongiano, 2022-02-02:
Hello friends, my name is Yamil Corco, I am from the city of Iquique in Chile(South America
), the first region of Tarapacá.
The APN configuration for Chile, the Entel company, if you can please put it in your database.

[connection]
id=cellular
type=gsm
autoconnect=true

[gsm]
apn=bam.entelpcs.cl

[ipv4]
method=auto

[ipv6]
addr-gen-mode=stable-privacy
method=auto

7 Nebra Outdoor hotspots belong to me in my small and beloved city, I have brought helium to this small place in the world, I have 7 hotspot and 1 belongs to a friend. and I have big plans for the people's internet.

seeding the internet of the people.

Greetings from Chile, my name is Yamil Corco alias yamilonigano el iquiqueño.

shawaj, 2022-02-02:
@yamilongiano thanks for the comment.

I've added this here https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/lte-apns.md#chile

closed 2022-02-02 by shawaj

yamilongiano, 2022-02-02:
sharing is the idea, greetings.

commit 90dc0d0a (2022-02-02): Add entel details

Closes #312